XGudron/UA3REO-DDC-Transceiver #22: Transformers specification

jgromes, 2022-02-28:
Hello everyone,

sorry if this is a dumb question or if it already has an answer somewhere - if it does, I was unable to find it. I'm attempting to build this transceiver. My version of the hardware might be slightly outdated, I ordered the PCBs quite a while ago, but didn't have the time to assemble until recently. The silkscreen has the label "rev27" for the motherboard, and "rev20" for the RF unit. I'm trying to find some infromation about some parts used, mainly the transformers:

* Motherboard: T3, T6
  * These have the same footprint as T4 (ADT4-6T), but I don't think they're the same, they're described as "1:1" instead of "1:4"
* RF board: TSWR, T3, T4, T5
  * I translated the instructions for TSWR from the schematic, but it's not really clear to me how to wind it
  * T3 is labeled as "8T M2000HM 10*6*5" in the schematic/BOM, but I was not able to find the exact part number anywhere. Does "8T" mean 8 turns and "10*6*5" the dimensions?

Thanks for any hints or instructions. I'm not really familiar with RF design, this is a bit of a soldering/radio excercise for me.

XGudron, 2022-02-28:
Hi!

T3 and T6 for example:
https://www.minicircuits.com/pdfs/TC1-1T+.pdf (the best)
https://www.minicircuits.com/pdfs/ADT1-6T+.pdf (works worse)

TSWR wind scheme:
http://dl2kq.de/ant/3-108-1.png
http://www.cqham.ru/forum/attachment.php?attachmentid=225833&d=1455452463
http://www.cqham.ru/forum/attachment.php?attachmentid=225828&d=1455451191

T3:
8-10 turns 2x0.5mm on 13.5х10х6.5 ferrite tube

T4:
2 parts: 5.5 turns RG-178 on Ferrite tube 13.5х10x6.5 each

*Ferrite tubes from old VGA cables, sizes don't really matter

T5: TC1-1T+ or ADT1-6T+

Good luck with the build!
We have a telegram channel where there are many participants and you can ask different questions about the assembly and share experience https://t.me/TRX_Wolf

jgromes, 2022-03-02:
@XGudron that clears it all up I think. Thank you very much for the instructions - and for the incredibly cool project :)

closed 2022-03-02 by jgromes