XGudron/UA3REO-DDC-Transceiver #33: add IF out

autodiag, 2023-07-08:
would it be possible to implement IF OUT output ? e.g. to the headphone source (L IF - P Audio like) or other? Basically a channel width of about 15kHz would be sufficient

XGudron, 2023-07-09:
The output of the IF is implemented via USB. For this, a separate device was created that outputs an IQ with a frequency of up to 48 kHz.

autodiag, 2023-07-09:
yes I know that but if I need to directly connect a third party decoder then the solution using only one cable is more elegant than a complicated solution via PC and it is not about IQ data but about pure analog IF

f4exb, 2023-07-24:
Hello @XGudron firstly congratulations for this awesome work. I have two questions regarding I/Q mode:

1. I haven't found in the menus the way to get the I/Q signal as a stereo R/L audio on the audio interface. Is there a specific setting somewhere? I am probably missing something. I use the latest firmware 8.1.0
2. Maybe I am missing something again but does the reverse path exist taking I/Q samples as R/L audio and use it to modulate the DUC? 

I am the principal developer of SDRangel (https://github.com/f4exb/sdrangel) and would love to interface the Wolf with this software for both Rx and Tx.

XGudron, 2023-07-27:
@f4exb Hello! SDRangel great project! I used it with HackRF when developing and debugging the Wolf. Thank you for your work !!!

To work with IQ directly, there is a special "IQ" mode, it allows you to both receive and transmit an IQ stream up to 48 kHz wide (both via an audio codec and via USB, if desired).
The button with its selection is not on the default screen, but this mode can be activated through the function buttons or by choosing a theme, where the mode is selected through a drop-down list.

XGudron, 2023-07-27:
@autodiag The IF in the transceiver has a 0Hz frequency (this is a direct conversion receiver). If you do not use IQ, then you will not be able to extract the useful part of the signal and demodulate the desired section. If you need an IQ output through an audio card, and not USB, then turn on the IQ mode

autodiag, 2023-07-27:
[image]

autodiag, 2023-07-27:
Well I thought to use the same circuit with ISL5857IAZ which uses for "fake" IF 10.7Mhz output ICOM IC-R8600..basic block wiring is more or less the same as with Wolf (filtered inputs ->ATT ->AD LTC2208-14> Altera IV -CPU for processing and control)
if you would like a detailed schematic for research I will send it to you by e-mail

Matin

StevenSaH, 2023-07-27:
take a closer look, they use a EP4CE55F23!
This is a FPGA with 484Pins and 56000LE, the FPGA used in the Wolf is a EP4CE22E22 with 144Pins and 22320 LE .. 
Take a closer look to the WolfMainboard, all IOs at the FPGA are in use

autodiag, 2023-07-27:
yes mate right the output deficit and other version of FPGA I did not realize..

f4exb, 2023-08-05:
Hello @XGudron I haven't been able to locate a function button related to I/Q mode. As far as I could check in the code they should all be listed here: https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/master/STM32/WOLF/Src/front_unit.c#L438

linoobs, 2023-08-05:
Just click "modularion" till i/q. PS. Use telegram for those questions: https://t.me/TRX_Wolf/615195 авг.

f4exb, 2023-08-05:
Unfortunately I am not fluent in Russian

f4exb, 2023-10-09:
Sorry to come back on this but I am still unable to activate the I/Q transmission mode. However I had a closer look at the code today... 

I noticed there was a TRX_MODE_IQ definition which I suppose governs the I/Q mode. Then here for some reason these lines are commented out: https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/master/STM32/WOLF/Src/lcd.c#L828 So I suppose the "IQ" mode icon was removed at some point and when in I/Q mode all mode icons are just turned off.

Tracking further TRX_MODE_IQ in the code I noticed that it was activated here: https://github.com/XGudron/UA3REO-DDC-Transceiver/blob/master/STM32/Src/trx_manager.c#L1440 This is in the BUTTONHANDLER_MODE_P function so I assume it handles the "Mode+" control. I guess it means that if I reach the SAM mode and press "Mode+" once again then I would turn on the I/Q mode. 

Are my assumptions correct? You may say that I could just try it out but I am asking for confirmation that I read the code correctly.

XGudron, 2023-10-15:
@f4exb Hello! I plan to add a choice of IQ mode for CN themes in the next release, I will solve this inconvenience