XGudron/UA3REO-DDC-Transceiver #34: A question about IF Output

Eguinane-Xor, 2023-07-16:
Hi

I didn't find any other way to contact you so i'm posting my question here :
Is it possible to redirect the signal through an IF Output port to a scanner "AOR AR-DV1" having an antenna IF Input to decode DMR, NXDN or even Icom D-Star ?

Kind regards.

f4exb, 2023-07-24:
If I am not mistaken the receiver is based on DDS (Direct Down Sampling) and therefore has no physical local oscillator.  Therefore there is no IF output possible. It would have t be reconstructed. The whole 0~61.44 MHz band is sampled and then the rest happens in the FPGA / Microcontroller in digital.

autodiag, 2023-07-25:
it is similarly solved for example ICOM IC-R8600 , which has a "fake" IF 10.7Mhz wide if I'm not mistaken 5mhz , derived from fpga

XGudron, 2023-07-27:
ICOM IC-R8600 has a hybrid structure with an external analog local oscillator, the result of which is already fed to the ADC. There are no superheterodyne circuits and a classical understanding of the IF in Wolk, all conversion takes place in digital form and at zero frequency