XGudron/UA3REO-DDC-Transceiver #36: Cant upgrade from version 6.8.0

jstanski, 2023-07-24:
Hi, when trying to upgrade from 6.8.0 to the latest version, I get an message saying no updates are available, digging into the code, the radio calls: ua3reo.ru/trx_services/check_fw_updates.php?...

The day after 6.8.0 was cut, Feb 7, a change was made in wifi.c to check updates from:
wolf-sdr.com/trx_services/check_fw_updates_2.php?...

Adding all the version info to the above urls in a web browser, the check_fw_updates.php call says there are no updates, while the newer check_fw_updates_2.php returns "new all".  It didn't matter whether ua3reo.ru or wolf-sdr.com was called.

Is it possible to have the older php file allow updates for version 6.8 as well?  Is there a hardware change in 7.0 that prevents this?

autodiag, 2023-07-25:
mean better use it direct to write SD card and use to update via to

XGudron, 2023-07-27:
You can update the transceiver to the latest version, there are no restrictions. It's just that in version 6.8.0 there is a problem due to which an incorrect FPGA update occurs, you need to step over this version manually

jstanski, 2023-08-01:
Thanks!  The odd thing is now the radio can see there's a new release when it couldn't before.  That was the original issue. 
However, the OTA download freezes half way.  I'm working towards manually placing binaries on the SD drive.

closed 2023-08-01 by jstanski