XGudron/UA3REO-DDC-Transceiver #40: Upgrading firmware

Starmike98, 2023-10-08:
I have looked everywhere but can’t find any instructions on how to upgrade the software ob my Wolf.  I am on version 6.8 and want to get up on version 8.  Can I get some help on this?

StevenSaH, 2023-10-08:
Short Boot0 
or use in Settings DFU Mode (but for this the Battiery must work), start the Wolf and use STM32Cube to flash the BIN File

StevenSaH, 2023-10-08:
look here for further Informations

https://github.com/XGudron/UA3REO-DDC-Transceiver/issues/36

Starmike98, 2023-10-08:
But how do you do DFU mode?   Do I hook something up to the USB port for that?   I am really a novice at this

StevenSaH, 2023-10-08:
Install STM32Cube and download the latest STM32 Firmware

https://www.dropbox.com/sh/jrs38c2iwtxk6c1/AACUw9yrOuv3P8LjwSxlcpqIa?dl=0

Install STM32Cube

install the Boot0 Jumper on the WolfMainboard
Power on the Wolf (you must  hold the Poweron Button the whole time during flashing!!)
your Computer should now see the Wolf in DFU Mode, press in STM32Cube in the top right corner connect, choose the bin file and upload it to the stm

StevenSaH, 2023-10-08:
There is a Telegram group for these types of questions.
You can get help there quicker than opening a case here

https://t.me/TRX_Wolf

Starmike98, 2023-10-08:
Thanks.  I am still really confused.  I looked at the Telegram chat but looks like it is all in Russian.  Sorry.  I am on version 6.8 so I should not choose the OTA update.  So if I open up the radio, take out the SD card, put it in my computer and put the new software on the card. Put the card back in the radio and then go though the updates from their.  Do you do STM first then FPGA or the other way

StevenSaH, 2023-10-08:
You can write there in english

on 6.8 flashing by SD Card could also fail! I dont know how big the selfupdate error was in 6.8. i could be only the FPGA update, but also the STM32 update!

When the STM32 Firmware file is on the SD choose
F8 than SD Card than FileManager, choose the BIN file and choose flash ..

When you changed the STM32 to a newer Version, you has to update the FPGA too.
When you has to do this manualy you need the USBBlaster on the JTAG connection, so be carefull with the updateprocess, and ask in the telegram channel if it is possible to do it on this way! i used stm32 and usb blaster to move from 6.8 to 7.0!

Starmike98, 2023-10-08:
I am on telegram and it is set to English bit all the post are still in Russian.  is there any way to translate them?

f4exb, 2023-10-09:
May be off topic but this is just to help you... You need to subscribe to Telegram premium to translate all messages at once. Else in the language menu enable the translate button then you can translate individual messages by right clicking on the message and choose "Translate".

XGudron, 2023-10-09:
https://github.com/XGudron/UA3REO-DDC-Transceiver/wiki/How-to-flash-a-Wolf-transceiver

Starmike98, 2023-10-09:
I got files over to the SD card. Started flashing FPGA.  Now I am getting EEPROM ERROR.  I was told to use St-Link to manually go in and flash.  Ordered an ST LINK device.  Is there any documentation on how to hook the st link device to the board on the Wolf?   Thanks

Starmike98, 2023-10-09:
I have also ordered the USB BLASTER.  Since it is the FPGA that is corrupted, I saw your procedure for using USB blaster to fix it.  Hopefully that will get me back up and running

Starmike98, 2023-10-10:
Got the USB Blaster installed and Quartus 13.1.  Computer sees the Blaster but can go any further.  Says the chain is broken.  Any hints to get me moving ahead

Starmike98, 2023-10-11:
I guess no one knows

Starmike98, 2023-10-11:
> https://github.com/XGudron/UA3REO-DDC-Transceiver/wiki/How-to-flash-a-Wolf-transceiver

This is the process I followed.  Thought it would be easy but it is not

StevenSaH, 2023-10-11:
It is easy... when you do exact the steps like in the howto it has to work.

Post a screenshot of your error, make a image of how you connect everthing

Starmike98, 2023-10-11:
Do you need to do the boot jumper for this or not.  How do you keep the radio from booting so you can flash

StevenSaH, 2023-10-11:
the boot0 jumper is only needed to programm the STM32

turn off the radio
connect the usb blaster
connect the usb blaster with the pc
turn on the radio (it will do a normal startup)
programm the fpga
turn off the radio
disconnect the usb blaster 
turn the radio on

Starmike98, 2023-10-11:
[image]
![IMG_0880](https://github.com/XGudron/UA3REO-DDC-Transceiver/assets/147280421/6feb916e-
[image]
[image]
[image]
[image]
[image]
74c5-4f43-b805-541040c88228)
[image]
[image]
[image]

Starmike98, 2023-10-11:
[image]
[image]
[image]
[image]

Starmike98, 2023-10-11:
Those are the steps I did.  Here are screenshots

closed 2023-10-11 by Starmike98

StevenSaH, 2023-10-11:
🤔
the radio was turned on? and no boot0 connection?

! When you close the boot0 jumper, the radio will not start, it only has power when you press and hold the power button, this is only for programming the STM !

you radio has to do a normal startup, but with connected USB Blaster. 

you error messages says that the fpga was not found, and could not accessed.

Starmike98, 2023-10-11:
Yes, radio is on.  Blank RED screen.  No boot0 jumper (I dont even know where that is at...lol), USB Blaster connected and powered by the computer when I turned on the radio

Starmike98, 2023-10-11:
Sorry to keep bugging you about this.  I went on telegram and they said to search the internet to find an original USB Blaster.  Like how would I know which is the original

Starmike98, 2023-10-11:
I order a different version from Amazon.  Wil be here tomorrow.  maybe something is wacky about the one i have

StevenSaH, 2023-10-11:
thats possible, but the red screen is not good..
is there more information about the error this.red screen shows??

Starmike98, 2023-10-11:
The RED screen first says
Loaded Default settings
then 
EEPROM Error
then 
Loaded default calibrations
then EEPROM Error (again)
Then the normal screen comes up briefly (which makes you think it is working) but spectrum part is blank
then goes to a blank RED screen

Starmike98, 2023-10-11:
This is a Chinese version

Starmike98, 2023-10-11:
[image]
[image]
[image]
[image]
[image]

StevenSaH, 2023-10-11:
Hmmm there is something wrong with the W25Q16 U7, but i could be because the fpga is not starting.

When the error is still there when you has updated the fpga, than you has to take a closer look

Starmike98, 2023-10-11:
[image]

Starmike98, 2023-10-11:
I assume the FPGA controls the spectrum scope so hat is why it is blank.  Nothing else comes up on the screen when i try to update FPGA

StevenSaH, 2023-10-11:
the stm gets some clocks from the fpga to access spi devices.
when the fpga does not work, this error could be possible, and your empty waterfall is a indicator that there is something wrong at the fpga

StevenSaH, 2023-10-11:
first try the other blaster
when this do es not.work replace the fpga

Starmike98, 2023-10-11:
Thanks for all the help

Starmike98, 2023-10-11:
Sorry for my ignorance, is the fpga a chip?  Where would i get one if i needed one?

StevenSaH, 2023-10-11:
yes it a chip

https://www.digikey.de/de/products/detail/intel/EP4CE22E22C8N/2288353

not easy to solder for a untrained person, maybe you can ask a ham to help you

StevenSaH, 2023-10-11:
Maybe  wd8bxs  can help you, he sometime builds mcHF for other hams in the US

https://www.digikey.de/de/products/detail/intel/EP4CE22E22C8N/2288353

Starmike98, 2023-10-12:
Hi, The new USB Blaster worked. Got the radio up and running. So then flashed STM form the SF card so they would be on the same version. After it flashed, the screen went BLACK. How long is it supposed to stay black?

This is what happened when I ORIGINALLY FLASH FPGA.  Screen went black.  So when I unplugged it from the power, it got the EEPROM Error.  So I am scared to touch it

Starmike98, 2023-10-12:
Now it will not power on, so it did not like version 8.1 STM.  I am not sure where to go from here