XGudron/UA3REO-DDC-Transceiver #41: FPGA Help

Starmike98, 2023-10-11:
I must have closed out my issue accidentally

Is there like a group of people that like to work on these radio (for pay of course) that I could send it to.  I can check on Telegram but most of those guys are overseas.  I am in the USA

Starmike98, 2023-10-11:
And I need the link again for FPGA chip

Starmike98, 2023-10-12:
The new USB Blaster worked.  Got the radio up and running.  So then flashed STM form the SF card so they would be on the same version.  After it flashed, the screen went BLACK.  How long is it supposed to stay black?

StevenSaH, 2023-10-13:
Normally it would show direct the boot logo.
Which STM32 file did you use? Your radio look like the 7" Version, so normaly the correct file is

WOLF-RA8875_GT911.bin

But try this first:
disconnect the Power
connect the power again
press and hold F1 and F8, and press power on

now you has to see "load default setting" and "load default calibrations"

release F1 and F8

after radio startup, press F8, to go into the menu
press and hold F8 until the calibration menu appears,
choose this menu and make your settings
When you has an SD Card with setting allready on it import these first via the SD Card menu.

an eeprom error mean that the stm cannoct access the eeprom.

when you used the wrong firmware file, no panic,
look in the HowTo how to flash the stm via STM32Cube, and flash it with the correct Version.
But you see errors ;) so i think your Firmwareversion was the correct one.

Starmike98, 2023-10-13:
I installed 8.1 FPGA via the USB blaster.  The radio then booted up fine, but STM32 was still on 6.8.  I had 8.1 on the SD CARD so flashed STM from there.  The radio shut off from there and would no longer power on.  I have ST LINK device so will try to instal 8.1 that way. Waiting for a cable to arrive to attach it to my computer.  Hope that works.  If I get this done I may never upgrade firmware again

StevenSaH, 2023-10-13:
You can connect a MicroUsbCable direct to the usb port on the motherboard.

Close the Boot0 Jumper on the mainboard,
press and hold the powerbutton the whole time!
The PC will install the driver (if stm32cube is installed) and stm32 will show you the connected device (maybe you have to rescan) and rhan you can upload the firmware.
After finish, release the power button. 
Open the Boot0Jumper, and try restart the radio

I dont know these chinese builds, i have build my by my own .. i never had problems like you, and i have flashed my wolf offen ;)

StevenSaH, 2023-10-13:
and use WOLF-RA8875_GT911.bin in the correct Version!
When you use FPGA FW v8.1 use STM FW v8.1
I could be that the last FPGA FW VersionNr is lower tgan STM FW, but when there is a FPGA FW with the same FW Version Nr  your stm will need at least this version!

ps i flash my wolf offen because i use always the latest dev-version an test ist ;)

Starmike98, 2023-10-13:
I agree. If I had the skills to build one I would have so I bought one from China.  I found the boot port but no pin.  Two holes, so I just out a wire across it  to jump the connection?

StevenSaH, 2023-10-13:
yes

Starmike98, 2023-10-13:
Thanks for all your help.  I did not have a cable that would go from the ST Link to the computer, so one is on the way.  Will be here this evening.  I will let you know how it works out

StevenSaH, 2023-10-13:
you can direct connect a MicroUSB Cable to the Mainboard, there must be a connection ;) for this way of flashing the stm32 there is no need of stm32link

StevenSaH, 2023-10-13:
below you find the "manual" i used two years before when i flashed my stm the first time

Programming the STM32 and the FPGA:
How to program the transceiver:

First you need an ST-Link-V2 for the STM32 and a USB blaster for the FPGA (for writing jic files for FPGA firmware) and BIN files (for STM32 firmware), which you can either compile yourself or find it in the Telegram Channel.

===STM32===
STM32 firmware can be built in 3 different ways:

1.The easiest way to program the STM is via the USB socket.

Connect the mainboard to the computer using a USB cable.
jumper the Boot0 jumper on the motherboard,
turn on the transceiver and press and hold the power button.
A driver may be required which then needs to be installed. (in the Soft/Driver folder)
Open the STM32CubeProg program,
select the detected DFU device and
select the firmware file (depends on the display used, e.g. WOLF-RA8875_GT911.bin) and start the programming process.
* release the power switch after completing the programming process,
* Remove USB cable.
* Remove the cable or jumper on the Boot0 jumper.
* The TRX can then be switched on and should boot up without any problems.
2. Programming with the ST-Link.

*If you haven't already done so, install the driver for the ST-Link
* Connect the ST-Link to the JTAG socket according to the circuit diagram
* Connect ST-Link to the computer using a USB cable
* Turn on the transceiver using the power button and press and hold the power button.
* Open the STM32CubeProg program,
* select the detected ST-LINK device and
Select the firmware file For the display used here, this file is used for programming: WOLF-RA8875_GT911.bin for the 7-inch display with GT911 and start the programming process.
* After completing programming, turn off the power to the TRX
* remove the connectors from the JTAG socket.
For the sake of completeness, the 3rd method:

with the ST-LINK programming device directly from the Keil programming environment,
To do this, after compiling the firmware, press F8 and wait until the process is completed.
The BIN file to be used for programming depends on the display used

Starmike98, 2023-10-13:
Thanks.  Is it normal for the radio to not power on after a problem with flashing

StevenSaH, 2023-10-13:
yes and no

deepends on how bad it goes wrong

when you press the powerbutton the STM gets power and sense on an input for the power on signaö and sets a Output that switch a MOSFET on the frontunit into "on" ... 
when you use the wrong firmware, this behavior could be wrong, or when the software in the stm is corrupt and will not start than the wolf will not switch on too

Starmike98, 2023-10-13:
I was so excited when the new USB blaster worked and I was up and running.  The joy quickly left when I flashed STM.   LOL

StevenSaH, 2023-10-13:
thuch behaviors could happen, but normaly you can not brick the STM, with ST/Link or via "boot0" you can allways send a new software to the STM

73

Starmike98, 2023-10-14:
Hi Steven

I got STM flash.  I never could get ST LINK to work so I hooked up the USB cable to the radio and the download happened. Radio comes up and works but do get an FPGA version error.  When I downloaded from Dropbox I got STM version 8.1,2.  For FPGA there was only 8.0 or 8.2 so I used 8.0.  Is there something else that can be done

StevenSaH, 2023-10-14:
FPGA v8.0 is correct for STM 8.1.2
could it that you installed STM 8.3.1 (the lastest)?
than you need FPGA v8.2

you can check the Version of the STM and  FPGA by press F8 and than choose "System Info"

but now you can use the SD Card to update the Firmware ;)

Starmike98, 2023-10-14:
Thanks for your patience with me and getting me through this.  And now I know how to do this if it happens again.

Starmike98, 2023-10-14:
Yes, FPGA was on 8.2. so I flash SYM from the SD card to 8.2  All is well

StevenSaH, 2023-10-14:
Top

now check your calibrations and your txco frequency and RFUnit settings, you can access it by press F8 for Menu and than press and hold F8 to activate the calibration menu 

and please dont you autocalibration ;) !! Use it only when you know the tx path is perfect. 
Do it manualy by setting rx gain for each band and adjust the fwd and bwd settings by mesaure it with a powermeter.

73

pt2nd, 2023-12-03:
Hi,
I'm new to the group and to this radio...
Had the same issue updating, but could get it back on with boot0 and usb cable.
The radio is turning on, reception is ok, but tuner and transmition not. Will get less than 5w on the meter and it is the 100w version.
The folder I downloaded contained all versions of STM32 for diferent models.
Already tried some of the bin files, radio will not work with some of them, but the ones that work are as I described.
Any way to find my model? Any other suggestions?
Thanks! Best 73!!

Noel - PT2ND

pt2nd, 2023-12-03:
> Did you also flash the latest FPGA after getting STM32 going?
> […](#)

Had not done that... thought it would have the previous one working.
So, just flashed FPGA from the sd card.
Noticed that I do get more power, but less than 50w nevertheless.
So I browsed through the calibration menu and found out that "RF Gain" was set to 40 in all hf bands. Just changed it to 100 and now the meter reads 100w when I tested...
I don't have an external meter at the moment to check if the reading is good.
Were my steps correct or did I chance some calibration in a way that the meter reads 100w but it actually is not?

Starting to get to know this radio at last!

Thanks!!

Starmike98, 2023-12-03:
If the FPGA is to far older the rig will sound dead.  Yes, you need to recalibrate after upgrading. I think it sets RF GAIN TO 40 but default.  100 is to high though. You will need an external meter to set the gain on each band. Use a CW signal and adjust the gain on each band until 100.  160 and 80m will not go to 100 so don’t push the gain to high.  10 and 12m will be similar.  I hope this helps

Starmike98, 2023-12-03:
One more thing, until you get an external meter, just set the gain to around 55. You don’t want to damage the finals

pt2nd, 2023-12-03:
Ok!
Thanks for all the tips!

Best 73s!