XGudron/UA3REO-DDC-Transceiver #51: no 144mhz power on RS-998

pt2nd, 2023-12-15:
Hello,

I've been using the RS-998 (7 inch screen 100w power unit) for a couple of weeks.
Had issues with firmware (flashed a wrong version and then had to recover), loosing all calibrations and settings.
I have done all calibration with a dummy load and external meter, everything seems fine in HF and 50mhz, with full 100w output, but I have noticed that I can't get any power out on 144mhz.
I tryed the auto-calibration and there's no power reading in the external meter. Changed RF gain manually and still nothing.
I have upgraded to this last 9.2.1 STM32 version and with the 8.2.0 FPGA.
Any suggestions on what to do?

Thanks!

Noel - PT2ND

StevenSaH, 2023-12-15:
In the Configuration you can choose your RF-Unit, check this first.

After that, if it still not work, check the DAC Output of the MB .. than the 2m PowerModule

F4IAH, 2024-01-02:
Good evening everyone.

I also have the RS998 and no power on 144mhz...
The RF unit is RU4PN....
Can you help me ?

73's Cédric F4IAH

StevenSaH, 2024-01-03:
Download the Schematics for the Mainboard and RF Board
1.Check your RF Gain in the calibation menu
2.test MB DAC Out for having a signal on TX
3.Check on RF Board the Signal before and after RD01 (maybe the RD01 is not working?)
4. Check the 2m PowerModule